JaapvanEkris/openrowingmonitor #59: Additional data at the end of TCX files

chrzr, 2024-10-27:
Hi!

First of all thanks to everyone involved in creating this great piece of open source software 👏
I got my first waterrower, a Joroto MR380, a few weeks ago and was able to retrofit it with ORM in just a few hours. 

I also wanted to automate the upload of my acitivities to Garmin Connect, so I created a service which uses inotifywait to trigger the conversion of the .tcx to a .fit file and also the upload to Garmin Connect.

The problem I am currently facing is that sometimes there are additional entries at the end of the tcx file. 

```
         <Trackpoint>
            <Time>2024-10-27T10:19:33.967Z</Time>
            <DistanceMeters>8442.17</DistanceMeters>
            <Cadence>25</Cadence>
            <Extensions>
              <ns2:TPX>
                <ns2:Speed>4.72</ns2:Speed>
                <ns2:Watts>294</ns2:Watts>
              </ns2:TPX>
            </Extensions>
            <HeartRateBpm>
              <Value>115</Value>
            </HeartRateBpm>
          </Trackpoint>
        </Track>
        <Extensions>
          <ns2:LX>
            <ns2:Steps>743</ns2:Steps>
            <ns2:AvgSpeed>4.69</ns2:AvgSpeed>
            <ns2:AvgWatts>291</ns2:AvgWatts>
            <ns2:MaxWatts>424</ns2:MaxWatts>
          </ns2:LX>
        </Extensions>
      </Lap>
      <Notes>Indoor Rowing, Drag factor: 32000.0 10-6 N*m*s2, Estimated VO2Max: UNDEFINED</Notes>
    </Activity>
  </Activities>
  <Author xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="Application_t">
    <Name>Open Rowing Monitor</Name>
    <Build>
      <Version>
        <VersionMajor>0</VersionMajor>
        <VersionMinor>9</VersionMinor>
        <BuildMajor>0</BuildMajor>
        <BuildMinor>0</BuildMinor>
      </Version>
      <LangID>en</LangID>
      <PartNumber>OPE-NROWI-NG</PartNumber>
    </Build>
  </Author>
</TrainingCenterDatabase>/Activities>
  <Author xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="Application_t">
    <Name>Open Rowing Monitor</Name>
    <Build>
      <Version>
        <VersionMajor>0</VersionMajor>
        <VersionMinor>9</VersionMinor>
        <BuildMajor>0</BuildMajor>
        <BuildMinor>0</BuildMinor>
      </Version>
      <LangID>en</LangID>
      <PartNumber>OPE-NROWI-NG</PartNumber>
    </Build>
  </Author>
</TrainingCenterDatabase>
```

I already tried looking for and adapting the corresponding code, which I found in WorkoutRecorder.js, but I just could not fix the issue. 
In the logging of openrowingmonitor service I see that there are multiple lines regarding the creation of tcx files which I suspect might be the problem here? 

> Oct 27 10:20:27 orm npm[5637]: saving session as RowingData file data/recordings/2024/10/2024-10-27_09-49-31_rowingData.csv...
> Oct 27 10:20:27 orm npm[5637]: saving session as tcx file data/recordings/2024/10/2024-10-27_09-49-31_rowing.tcx...
> Oct 27 10:20:27 orm npm[5637]: saving session as tcx file data/recordings/2024/10/2024-10-27_09-49-31_rowing.tcx...
> Oct 27 10:20:27 orm npm[5637]: saving session as tcx file data/recordings/2024/10/2024-10-27_09-49-31_rowing.tcx...
> Oct 27 10:20:37 orm ORMWatchdog.sh[2156]: /opt/openrowingmonitor/data/recordings/2024/10/2024-10-27_09-49-31_rowing.tcx

Could it be that the corresponding function is called multiple times in parallel causing the additional entries in the tcx?

Any ideas would be appreciated! 

Kind Regards,
Christoph

JaapvanEkris, 2024-10-27:
It indeed generates additional data: the recovery heart rate is measured every minute (max 3 times) and added to the remarks field for sessions with a HR above 70% MaxHR. I use a simple bash polling for new files in a folder that are older than 3 minutes.

chrzr, 2024-10-27:
Thanks for the quick reply!
Ok so this is intended behaviour - but what still confuses me a bit and what is also seems to cause the problems with conversion is the following duplicate data at the end of the tcx file (what i meant by additional, probably phrased wrong) 

```
  </Activities>
  <Author xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="Application_t">
    <Name>Open Rowing Monitor</Name>
    <Build>
      <Version>
        <VersionMajor>0</VersionMajor>
        <VersionMinor>9</VersionMinor>
        <BuildMajor>0</BuildMajor>
        <BuildMinor>0</BuildMinor>
      </Version>
      <LangID>en</LangID>
      <PartNumber>OPE-NROWI-NG</PartNumber>
    </Build>
  </Author>
</TrainingCenterDatabase>/Activities>
  <Author xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="Application_t">
    <Name>Open Rowing Monitor</Name>
    <Build>
      <Version>
        <VersionMajor>0</VersionMajor>
        <VersionMinor>9</VersionMinor>
        <BuildMajor>0</BuildMajor>
        <BuildMinor>0</BuildMinor>
      </Version>
      <LangID>en</LangID>
      <PartNumber>OPE-NROWI-NG</PartNumber>
    </Build>
  </Author>
</TrainingCenterDatabase>
```

Using the tcx directly works - but I also sometimes had to manually remove the last part of the tcx for Garmin Connect to import the file 🤔
I also like that it converts the data to be more rowing specific (strokes, strokes rate, pace etc.)

JaapvanEkris, 2024-10-27:
> Thanks for the quick reply! Ok so this is intended behaviour - but what still confuses me a bit and what also seems to cause the problems with conversion is the following duplicate data at the end of the tcx file (what i meant by additional, probably phrased wrong)

This is just plain weird behaviour. In my current setup (which uses Architecture_Revision, but the core tcx writer is the same) I don't see this behaviour. 

> I tried commenting out the code where the Author data is generated, but then I just get the data above as duplicate, which is why I thought that it must have something to do with parallel execution of the function.

There is a minute in between. That should be plenty of time to recalculate and write any data. Might be that you keep grabbing the file as soon as it is written. Perhaps give it a bit more time to write?

> Using the tcx directly works - but I also sometimes had to manually remove the last part of the tcx for Garmin Connect to import the file 🤔 I also like that it converts the data to be more rowing specific (strokes, strokes rate, pace etc.)

Changing the type of session to "Indoor Rowing" always did that trick for me. Another (easier) approach is to see if your Garmin watch supports the indoor rowing ANT+ profile and add a simple ANT+ dongle to ORM. That actually adds data as not only strokerate gets picked up, but on higher end watches also the distance per stroke and breathing.

chrzr, 2024-10-27:
Ok so the tcx file is only generated after pressing the reset/end session button?
My script indeed only waits 10 seconds after the file is created before triggering the conversion, so I'll up that to 120 seconds just to be safe.

What is weird is that I have older tcx files where I didn't have any watchdog script running and there I see the same duplicate data - maybe I'll try and setup a second SD card with a fresh install to see if it shows the same behaviour 🤔

I actually have a Garmin Instinct 2X and I already have an ANT+ dongle connected for HR monitoring - I'll have to check if my watch supports it. That would indeed make things a bit easier :)

JaapvanEkris, 2024-10-27:
> Ok so the tcx file is only generated after pressing the reset/end session button?

It is also generated on a stop or pause.

> My script indeed only waits 10 seconds after the file is created before triggering the conversion, so I'll up that to 120 seconds just to be safe.

Yeah, although your next paragraph suggests it isn't due to a too quick pickup.

> What is weird is that I have older tcx files where I didn't have any watchdog script running and there I see the same duplicate data - maybe I'll try and setup a second SD card with a fresh install to see if it shows the same behaviour 🤔

In the new architecture it is nicely separated. I am considering a compkete rewrite of the tcx recorder as its structure isn't easy to read/understand. I am still thinking about adding a native fit-file writer as it can contain more data than tcx, but it isn't the easiest format to create.

> I actually have a Garmin Instinct 2X and I already have an ANT+ dongle connected for HR monitoring - I'll have to check if my watch supports it. That would indeed make things a bit easier :)

That is way cooler if it works. Switched to that approach when @Abasz made it, never looked back.

JaapvanEkris, 2024-10-28:
> > I am still thinking about adding a native fit-file writer as it can contain more data than tcx, but it isn't the easiest format to create.
> 
> Tha javascript stack of fit SDK does not support writing of fit files only reading.... I mean it's possible to implement, probably there are libraries too.

There is a new one, which seems to work: https://github.com/markw65/fit-file-writer

My approach is to rewrite the tcx recorder first and add the layers in the data in the code as well (i.e. header, workout, laps, position reports) and test that thoroughly. When that works, hopefully the fit-writer will be mature enough to be integrated on the similar structure.

JaapvanEkris, 2024-11-08:
Just an update from my side: I have an initial version of the fit-file writer working, and its files are accepted by Garmin and Strava (that part actually took the most time...). In the weekend I'll do some additional testing to see how it behaves with pauses and intervals, and then I'll put it in the Architecture_Revision branch.

Please note, to install the needed npm you need to do a "npm install @markw65/fit-file-writer --save" in the /opt/openrowingmonitor directory by hand. I'll update the package list as well, as soon as I found an easy way of doing that, but I'm not certain how our updatescript behaves when new packages are included.

Abasz, 2024-11-09:
oooh I loved FIT SDK... 

The distance per stroke is actually horribly documented. First of all it has different names for the lap (avg) and for records, not to mention that for the records it was only made public a few years ago after I filed a [bug report](https://forums.garmin.com/developer/fit-sdk/f/discussion/323579/missing-stroke_distance-field-on-the-record-message)

It is called `cycle_length16` for the record and avg_stroke_distance for the lap and session.

JaapvanEkris, 2024-11-11:
Guys, thanks to the great help from the developer of Fit-file-writer, the architecture_review branch now contains a fit-recorder. You need to activate it by adding the settings 'createFitFiles' and 'gzipFitFiles'.

I have updated the package list, but if the package is missing (I don't know if it is automatically refreshed in the update script), you can fix it with
```
npm install @markw65/fit-file-writer --save
```
Also many thanks to @Abasz, as his forum post convinced Mark that Garmin has some documentation issues....

closed 2024-11-11 by JaapvanEkris

Abasz, 2024-11-11:
> Come to think of it, @Abasz , do you know if there is an assumed hierarchical relation between laps and records? Now, similar to the tcx file which has hierarchy, I add records for the start and end of the lap. But why would I do that, as they are completely separated lists?

The fit messages are separate entities actually (this is an over statement as certain messages can be linked to others like the descriptors). This also means that order does not matter. The fit file is written in the order the events occur. 

https://developer.garmin.com/fit/cookbook/decoding-activity-files/

If you look at the developer guide example, matching records to sessions and laps is done via the time stamp:

_"If there is only one Session message in the file, then all messages in the file belong to that single session. If there are multiple Session messages in the file, then messages are considered to be part of a session if their timestamp occurs within the time span of that session."_

On one hand this makes life easy for the writing, on the other hand it makes life hell when you try to parse it back :)

Even more annoyingly they have decided to use events for example the C# parsing. On one hand this is quick and dirty solution as it liberates Garmin from caring about things like when a recovery heart rate event gets between two records (i.e. nothing is continuous) on the other hand parsing requires whole file before you can start making sense of it. I mean they are quite open about this choice so I dont blame them :)