JaapvanEkris/openrowingmonitor #62: webbrowserkiosk failure

KjellVerb, 2024-11-04:
I tried everything including starting from a fresh install but still the screen will not work. Here is the log of the webbrowserkiosk service, can anyone give further suggestions to try? Should I not have installed the RPI OS Lite but the full instead?

```
11/4/2024, 10:34:52 PM | xinit[25755] | X.Org X Server 1.21.1.7
11/4/2024, 10:34:52 PM | xinit[25755] | X Protocol Version 11, Revision 0
11/4/2024, 10:34:52 PM | xinit[25755] | Current Operating System: Linux rowingmonitor 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l
11/4/2024, 10:34:52 PM | xinit[25755] | Kernel  command line: coherent_pool=1M 8250.nr_uarts=0  snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1  snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0   vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000   console=ttyS0,115200 console=tty1 root=PARTUUID=51870bf6-02  rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=BE
11/4/2024, 10:34:52 PM | xinit[25755] | xorg-server 2:21.1.7-3+rpt3+deb12u7 (https://www.debian.org/support)
11/4/2024, 10:34:52 PM | xinit[25755] | Current version of pixman: 0.42.2
11/4/2024, 10:34:52 PM | xinit[25755] | Before reporting problems, check http://wiki.x.org
11/4/2024, 10:34:52 PM | xinit[25755] | to make sure that you have the latest version.
11/4/2024, 10:34:52 PM | xinit[25755] | Markers: (--) probed, (**) from config file, (==) default setting,
11/4/2024, 10:34:52 PM | xinit[25755] | (++) from command line, (!!) notice, (II) informational,
11/4/2024, 10:34:52 PM | xinit[25755] | (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
11/4/2024, 10:34:52 PM | xinit[25755] | (==) Log file: "/var/log/Xorg.0.log", Time: Mon Nov  4 22:34:52 2024
11/4/2024, 10:34:52 PM | xinit[25755] | (==) Using system config directory "/usr/share/X11/xorg.conf.d"
11/4/2024, 10:34:57 PM | xinit[25770] | sed: can't read /home/pi/.config/chromium/Local State: No such file or directory
11/4/2024, 10:34:57 PM | xinit[25771] | sed: can't read /home/pi/.config/chromium/Default/Preferences: No such file or directory
11/4/2024, 10:35:00 PM | xinit[25769] | Openbox-Message: Unable to find a valid menu file "/var/lib/openbox/debian-menu.xml"
11/4/2024, 10:35:00 PM | xinit[25812] | ERROR: openbox-xdg-autostart requires PyXDG to be installed
11/4/2024, 10:35:04 PM | xinit[25772] | [25772:25824:1104/223504.170294:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:04 PM | xinit[25772] | [25772:25824:1104/223504.173153:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:04 PM | xinit[25772] | [25772:25824:1104/223504.173433:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:04 PM | xinit[25772] | [25772:25824:1104/223504.173635:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:04 PM | xinit[25772] | [25772:25824:1104/223504.173809:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:05 PM | xinit[25772] | [25772:25824:1104/223505.457945:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:05 PM | xinit[25772] | [25772:25824:1104/223505.458254:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:06 PM | xinit[25772] | [25772:25820:1104/223506.253935:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:06 PM | xinit[25772] | [25772:25824:1104/223506.577297:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:06 PM | xinit[25772] | [25772:25824:1104/223506.577548:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25772:1104/223507.385110:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path=  /org/freedesktop/DBus: unknown error type:
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25822:1104/223507.386982:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25882:1104/223507.578944:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.DBus.Properties.Get:  object_path= /org/freedesktop/UPower:  org.freedesktop.DBus.Error.ServiceUnknown: The name  org.freedesktop.UPower was not provided by any .service files
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25882:1104/223507.584513:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.UPower.GetDisplayDevice:  object_path= /org/freedesktop/UPower:  org.freedesktop.DBus.Error.ServiceUnknown: The name  org.freedesktop.UPower was not provided by any .service files
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25882:1104/223507.587599:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.UPower.EnumerateDevices:  object_path= /org/freedesktop/UPower:  org.freedesktop.DBus.Error.ServiceUnknown: The name  org.freedesktop.UPower was not provided by any .service files
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25772:1104/223507.614545:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path=  /org/freedesktop/DBus: unknown error type:
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25833:1104/223507.615895:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25772:1104/223507.709595:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path=  /org/freedesktop/DBus: unknown error type:
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25833:1104/223507.711637:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25772:1104/223507.745707:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path=  /org/freedesktop/DBus: unknown error type:
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25833:1104/223507.746676:ERROR:bus.cc(407)]  Failed to connect to the bus: Could not parse server address: Unknown  address type (examples of valid types are "tcp" and on UNIX "unix")
11/4/2024, 10:35:07 PM | xinit[25772] | [25772:25772:1104/223507.828503:ERROR:object_proxy.cc(576)]  Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path=  /org/freedesktop/DBus: unknown error type:
11/4/2024, 10:35:08 PM | xinit[25837] | [25837:25891:1104/223508.925783:ERROR:v4l2_utils.cc(584)] Could not open /dev/video10: Permission denied (13)
11/4/2024, 10:35:09 PM | xinit[25837] | [25837:25891:1104/223509.196842:ERROR:v4l2_utils.cc(584)] Could not open /dev/video10: Permission denied (13)
```

KjellVerb, 2024-11-04:
By the way the files that give "No such file or directory" very much exist so I don't understand that error at all.
```
pi@openrowing:~ $ ls /home/pi/.config/chromium/Default/Prefere*
/home/pi/.config/chromium/Default/Preferences
pi@openrowing:~ $ ls /home/pi/.config/chromium/Local\ State
'/home/pi/.config/chromium/Local State'
```

KjellVerb, 2024-11-04:
After trying the same on raspberry OS (the non-Lite) this is the result:
```
○ webbrowserkiosk.service - X11 Web Browser Kiosk
     Loaded: loaded (/lib/systemd/system/webbrowserkiosk.service; enabled; preset: enabled)
     Active: inactive (dead) since Mon 2024-11-04 23:58:43 CET; 2min 59s ago
   Duration: 1min 37.787s
    Process: 1188 ExecStart=xinit /opt/openrowingmonitor/install/webbrowserkiosk.sh -- -nocursor (code=exited, status=0/SUCCESS)
   Main PID: 1188 (code=exited, status=0/SUCCESS)
        CPU: 87ms

Nov 04 23:57:06 rowingmonitor xinit[1191]:         If this server is no longer running, remove /tmp/.X0-lock
Nov 04 23:57:06 rowingmonitor xinit[1191]:         and start again.
Nov 04 23:57:06 rowingmonitor xinit[1191]: (EE)
Nov 04 23:57:06 rowingmonitor xinit[1191]: (EE)
Nov 04 23:57:06 rowingmonitor xinit[1191]: Please consult the The X.Org Foundation support
Nov 04 23:57:06 rowingmonitor xinit[1191]:          at http://wiki.x.org
Nov 04 23:57:06 rowingmonitor xinit[1191]:  for help.
Nov 04 23:57:06 rowingmonitor xinit[1191]: (EE)
Nov 04 23:58:43 rowingmonitor xinit[1507]: X connection to :0 broken (explicit kill or server shutdown).
Nov 04 23:58:43 rowingmonitor systemd[1]: webbrowserkiosk.service: Deactivated successfully.

```

JaapvanEkris, 2024-11-06:
What OS version did you install?

KjellVerb, 2024-11-06:
RPi OS Bookworm, tried all 4 versions between 64/32 bit and Lite/Full.
It seems like a chromium issue. After some manual installation and editing of the webbrowserkiosk.sh it now works by using Firefox as the kiosk browser app.

JaapvanEkris, 2024-11-06:
> RPi OS Bookworm, tried all 4 versions between 64/32 bit and Lite/Full. It seems like a chromium issue.

Might this be related to this change in RPi Bookworm:
https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/ ?

KjellVerb, 2024-11-08:
> Might this be related to this change in RPi Bookworm: https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/ ?

I'm not sure because I changed my PI each time to use X via `raspi-config`

JaapvanEkris, 2024-12-07:
I did a clean fresh install of Bookworm a couple of times now, but I didn't have any issues. Everything installed perfectly.

JaapvanEkris, 2024-12-16:
As the poster hasn't responded in a week, and I can't replicate this, I'm going to close it. Will reopen if this issue pops up again.

closed 2024-12-16 by JaapvanEkris