JaapvanEkris/openrowingmonitor #76: Update of ORM branch "V1Beta-Architecture_Revision" fails with "Error: Cannot find module '../lightningcss.linux-arm-gnueabihf.node'"

Gordon-Shumway2, 2025-01-22:
Hi, 

**summary**:  a previously working installation of branch "V1Beta-Architecture_Revision" on Pi Zero 2 fails to start after an update attempt via "updateopenrowingmonitor.sh" with "Error: Cannot find module '../lightningcss.linux-arm-gnueabihf.node'"

**Longer version** (with log file excerpts at the end of this post): 

ORM branch "V1Beta-Architecture_Revision" was working fine last night (21.01.2025). Today I ran "updateopenrowingmonitor.sh" and it returned an error: 
 

> [!] Error: Cannot find module '../lightningcss.linux-arm-gnueabihf.node'

Starting ORM now fails and returns this error:
---

> Jan 22 12:33:41 pirower npm[2305]: > node app/server.js
> Jan 22 12:33:45 pirower npm[2319]: file:///opt/openrowingmonitor/app/tools/ConfigManager.js:33
> Jan 22 12:33:45 pirower npm[2319]:   if (Object.keys(configToCheck.rowerSettings).includes('minumumForceBeforeStroke')) {
> Jan 22 12:33:45 pirower npm[2319]:              ^
> Jan 22 12:33:45 pirower npm[2319]: TypeError: Cannot convert undefined or null to object
> Jan 22 12:33:45 pirower npm[2319]:     at Function.keys (<anonymous>)
> Jan 22 12:33:45 pirower npm[2319]:     at runConfigMigration (file:///opt/openrowingmonitor/app/tools/ConfigManager.js:33:14)
> Jan 22 12:33:45 pirower npm[2319]:     at file:///opt/openrowingmonitor/app/tools/ConfigManager.js:111:1

---

I believe that the startup error is due to a typo in a variable name - note the 'min**u**mumForceBeforeStroke' rather than 'min**i**mumForceBeforeStroke'. Wasn't this a bug some versions ago which was fixed? 

As to the initial update error: nothing really changed on my system between last night and today, so this rather looks like an npm requirement which is not covered by the update script.

Relevant log file excerpts: 

--- final lines of "updateopenrowingmonitor.sh"  ---

> npm warn deprecated eslint@8.57.1: This version is no longer supported. Please see https://eslint.org/version-support for other options.
> 
> added 782 packages, and audited 783 packages in 14m
> 
> 157 packages are looking for funding
>   run `npm fund` for details
> 
> 11 vulnerabilities (6 moderate, 5 high)
> 
> To address all issues (including breaking changes), run:
>   npm audit fix --force
> 
> Run `npm audit` for details.
> 
> > openrowingmonitor@0.9.0 build
> > rollup -c
> 
> [!] Error: Cannot find module '../lightningcss.linux-arm-gnueabihf.node'
> Require stack:
> - /opt/openrowingmonitor/node_modules/lightningcss/node/index.js
> - /opt/openrowingmonitor/node_modules/@web/rollup-plugin-html/dist/output/emitAssets.js
> - /opt/openrowingmonitor/node_modules/@web/rollup-plugin-html/dist/rollupPluginHTML.js
> - /opt/openrowingmonitor/node_modules/@web/rollup-plugin-html/dist/index.js
>     at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1225:15)
>     at Function.Module._load (node:internal/modules/cjs/loader:1051:27)
>     at Module.require (node:internal/modules/cjs/loader:1311:19)
>     at require (node:internal/modules/helpers:179:18)
>     at Object.<anonymous> (/opt/openrowingmonitor/node_modules/lightningcss/node/index.js:21:22)
>     at Module._compile (node:internal/modules/cjs/loader:1469:14)
>     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1548:10)
>     at Module.load (node:internal/modules/cjs/loader:1288:32)
>     at Function.Module._load (node:internal/modules/cjs/loader:1104:12)
>     at Module.require (node:internal/modules/cjs/loader:1311:19)

---  end of final lines of "updateopenrowingmonitor.sh"  ---

A subsequent "updateopenrowingmonitor.sh" returns

---

> root@pirower:~# updateopenrowingmonitor.sh
> Update script for Open Rowing Monitor
> 
> Checking for new version...
> You are using the latest version of Open Rowing Monitor from branch "V1Beta-Architecture_Revision".

----

Restarting ORM via "systemctl" fails:

---   log of ORM start ----

> root@pirower:~#  systemctl restart openrowingmonitor;logorm
> -- Journal begins at Tue 2023-12-05 03:57:17 CET. --
> Jan 22 11:33:33 pirower npm[609]: OpenRowingMonitor shutdown at 11:33, at 0.00000 seconds,distance 0.0m
> Jan 22 11:33:33 pirower systemd[1]: Stopping Open Rowing Monitor...
> Jan 22 11:33:33 pirower npm[609]: fit file has not been written, as there were not enough strokes recorded (minimum 10 seconds and two strokes)
> Jan 22 11:33:33 pirower npm[623]: 2025-01-22 11:33:33 sigHandler: Unhandled signal 18, terminating
> Jan 22 11:33:33 pirower npm[623]: 2025-01-22 11:33:33 sigHandler: Unhandled signal 11, terminating
> Jan 22 11:33:33 pirower npm[623]: 2025-01-22 11:33:33 sigHandler: Unhandled signal 11, terminating
> Jan 22 11:33:33 pirower systemd[1]: openrowingmonitor.service: Succeeded.
> Jan 22 11:33:33 pirower systemd[1]: Stopped Open Rowing Monitor.
> Jan 22 11:33:33 pirower systemd[1]: openrowingmonitor.service: Consumed 20.889s CPU time.
> Jan 22 12:33:40 pirower systemd[1]: Started Open Rowing Monitor.
> Jan 22 12:33:41 pirower npm[2305]: > openrowingmonitor@0.9.0 start
> Jan 22 12:33:41 pirower npm[2305]: > node app/server.js
> Jan 22 12:33:45 pirower npm[2319]: file:///opt/openrowingmonitor/app/tools/ConfigManager.js:33
> Jan 22 12:33:45 pirower npm[2319]:   if (Object.keys(configToCheck.rowerSettings).includes('minumumForceBeforeStroke')) {
> Jan 22 12:33:45 pirower npm[2319]:              ^
> Jan 22 12:33:45 pirower npm[2319]: TypeError: Cannot convert undefined or null to object
> Jan 22 12:33:45 pirower npm[2319]:     at Function.keys (<anonymous>)
> Jan 22 12:33:45 pirower npm[2319]:     at runConfigMigration (file:///opt/openrowingmonitor/app/tools/ConfigManager.js:33:14)
> Jan 22 12:33:45 pirower npm[2319]:     at file:///opt/openrowingmonitor/app/tools/ConfigManager.js:111:1
> Jan 22 12:33:45 pirower npm[2319]: Node.js v20.18.0
> Jan 22 12:33:45 pirower systemd[1]: openrowingmonitor.service: Main process exited, code=exited, status=1/FAILURE
> Jan 22 12:33:45 pirower systemd[1]: openrowingmonitor.service: Failed with result 'exit-code'.
> Jan 22 12:33:45 pirower systemd[1]: openrowingmonitor.service: Consumed 4.869s CPU time.
> Jan 22 12:33:45 pirower systemd[1]: openrowingmonitor.service: Scheduled restart job, restart counter is at 1.
> Jan 22 12:33:45 pirower systemd[1]: Stopped Open Rowing Monitor.
> Jan 22 12:33:45 pirower systemd[1]: openrowingmonitor.service: Consumed 4.869s CPU time.
> Jan 22 12:33:45 pirower systemd[1]: Started Open Rowing Monitor.

(... rest of logs show the same error stack repeatedly)
--- end of ORM start ---

Kind regards, 
Michael

JaapvanEkris, 2025-01-22:
> ORM branch "V1Beta-Architecture_Revision" was working fine last night (21.01.2025). Today I ran "updateopenrowingmonitor.sh" and it returned an error:
> 
> > [!] Error: Cannot find module '../lightningcss.linux-arm-gnueabihf.node'

When I do a clean install it is found and installed.

I notice that sometimes npm reacts pretty badly to network issues. So a failed install could be caused by this. You can resolve this by switching to the app directory by executing `cd /opt/openrowingmonitor` and then manually do a `sudo npm install`. This will all install all npm dependencies. If the error persists, you could remove the 'package-lock.json' and again do a `sudo npm install` to force a fresh install of all packages.

As the package that is instaled is part of the rollup package, yoou'll need to do a `sudo npm run build` after the install of all packages (in the same directory). This will execute the rollup for you (this command takes about three minutes to execute on a Pi4).
 
> ## Starting ORM now fails and returns this error:
> > Jan 22 12:33:41 pirower npm[2305]: > node app/server.js
> > Jan 22 12:33:45 pirower npm[2319]: file:///opt/openrowingmonitor/app/tools/ConfigManager.js:33
> > Jan 22 12:33:45 pirower npm[2319]:   if (Object.keys(configToCheck.rowerSettings).includes('minumumForceBeforeStroke')) {
> > Jan 22 12:33:45 pirower npm[2319]:              ^
> > Jan 22 12:33:45 pirower npm[2319]: TypeError: Cannot convert undefined or null to object
> > Jan 22 12:33:45 pirower npm[2319]:     at Function.keys ()
> > Jan 22 12:33:45 pirower npm[2319]:     at runConfigMigration (file:///opt/openrowingmonitor/app/tools/ConfigManager.js:33:14)
> > Jan 22 12:33:45 pirower npm[2319]:     at file:///opt/openrowingmonitor/app/tools/ConfigManager.js:111:1
> 
> I believe that the startup error is due to a typo in a variable name - note the 'min**u**mumForceBeforeStroke' rather than 'min**i**mumForceBeforeStroke'. Wasn't this a bug some versions ago which was fixed?

This specific line actually checks for the presence of the misspelled version and alerts the user to this problem (and fixes it). So it is supposed to be there.
 
> > added 782 packages, and audited 783 packages in 14m
> > 157 packages are looking for funding
> > run `npm fund` for details
> > 11 vulnerabilities (6 moderate, 5 high)
> > To address all issues (including breaking changes), run:
> > npm audit fix --force
> > Run `npm audit` for details.

This means that according to npm all packages were installed correctly. This should have installed the lightningcss.linux-arm-gnueabihf as it is listed in the package-lock.json as a dependency for rollup

> openrowingmonitor@0.9.0 build
> rollup -c
>
>
> [!] Error: Cannot find module '../lightningcss.linux-arm-gnueabihf.node'
> Require stack:
>
> * /opt/openrowingmonitor/node_modules/lightningcss/node/index.js
> * [...]

This implies that rollup has failed, and thus the compressed version of index.js isn't made. And thus openrowingmonitor will not start, leading to your errors.

Hope this helps.

Gordon-Shumway2, 2025-01-23:
Hello Jaap, 

thank you for looking into this so fast! I had a closer look at my configuration and noticed that within _config.js_ the _rowerSettings_ key pointed to a non-existing entry within _rowerProfiles.js_ - and that seems to have triggered the error message about 
     

>  if (Object.keys(configToCheck.rowerSettings).includes('minumumForceBeforeStroke')) {

Apparently during some brain freeze I removed the section about my NeonProV rower within _rowerProfiles.js_ but left the reference to it in _config.js_. That caused the first issue which went away once I added the NeonProV-specific section to _rowerProfile.js_. 

The other issue about missing module **lightningcss.linux-arm-gnueabihf.node** did not show up anymore once I had run "npm install". 

I now have a working ORM - thank you!

Kind regards, 
Michael

JaapvanEkris, 2025-02-08:
The workaround is now in the main branch. As stoprocent hasn't responded in over a week, I'm assuming no fix will come from his branch of NoBle/Bleno. So for the long term, I guess we have to move to another package?

closed 2025-02-08 by JaapvanEkris