JaapvanEkris/openrowingmonitor #81: placeholders physics docs

jrhunger, 2025-02-09:
In docs/physics_openrowingmonitor.md there is a section that reads as follows, where it seems like there are multiple conflicting uses of @@ that were intended to be replaced by other things, thought it's hard for me to guess what and open a pull request.

---
* For approach 1, on a Concept2, the typical R<sup>2</sup> is around @@ (low drag) to @@ (high drag) for steady state rowing. Also, calculated drag has a standard deviation of @@ (low drag) to @@ (high drag) across a session.
* For approach 2, On a Concept2, the typical R<sup>2</sup> is around 0.96 (low drag) to 0.99 (high drag) for steady state rowing. Also, calculated drag has a standard deviation of @@ (low drag) to @@ (high drag) across an entire session.

Therefore, we choose @@

JaapvanEkris, 2025-02-09:
Yeah, that one is on me. I'm stil conducting these tests, but fixing the Bluetooth connection issues took precedence here (which required use of the same test system). Next release will include that data.

jrhunger, 2025-02-09:
Thanks for forking and evolving this repo including the attention paid to the docs! I have been making a boutique/personal ESP32-based system for my NordicTrak rower mostly for the fun of doing it, but reading through all the docs on the other projects out there as I add new features (currently trying to get drag detection).  Maybe I will hook up a pi and try this project out too.

Abasz, 2025-02-09:
We have a port of this for esp32 called ESP rowing monitor: https://github.com/Abasz/ESPRowingMonitor

It uses the same robust rowing engine (with some minimal tweaks) supports FTMS ble, and has a gui that is purely based on ble (including force curves too).

Check it out if you would like.

JaapvanEkris, 2025-02-09:
> Thanks for forking and evolving this repo including the attention paid to the docs! I have been making a boutique/personal ESP32-based system for my NordicTrak rower mostly for the fun of doing it, [...]

On a Pi, we are years ahead of others. We learned a lot from other implementations, but their approach was too simple to use reliably. I owned a NordicTrack as well, and their flywheel was too wobbly/unbalanced to use conventional numerical approaches used by anyone else. Other machines have similar issues. So our regression based approach is robust and delivers good metrics even with poor data quality, but is CPU intensive.

This is where the great work of @Abasz comes in. Where this project has quite a luxurious position in terms of CPU cycles, and thus can apply extremely robust math, the ESP lacks that CPU power. So Abasz has to be smarter, knowing where to save these precious CPU cycles while not hurting metrics too much, is where Abasz' solution really made some big advances. So I would really recommend looking at his project (and contributing).

closed 2025-07-06 by JaapvanEkris