JaapvanEkris/openrowingmonitor #90: Document the data object with JS Doc

Abasz, 2025-03-18:
Defining the `metrics` and `metricsContext` as a JS Doc annotation would help knowing what properties are available on these object which would help a lot on the consumer side as VSCode and other IDEs would be able to provide inteli-sense and type checking.

We can think about what other things should be documented. I have these in mind so far:
1. Commands
2. Certain events
3. Rower states

JaapvanEkris, 2025-03-18:
> Defining the `metrics` and `metricsContext` as a JS Doc annotation would help knowing what properties are available on these object which would help a lot on the consumer side as VSCode and other IDEs would be able to provide inteli-sense and type checking.

That would be a good idea. I'll look into it, but will introduce it gradually in my code.

> We can think about what other things should be documented. I have these in mind so far:
> 
> 1. Commands

Commands between the managers are documented in the architecture document, see https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/Architecture.md#command-flow

More detailed info could/should indeed be standardised in the code itself, and JSDoc might be a good way to do it.

> 3. Certain events
Flags in the metricscontext are documented in https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/Architecture.md#rowing-metrics-flow

> 4. Rower states

Session states are documented here: https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/Architecture.md#sessionmanagerjs
underlying Rower states are documented here: https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/Architecture.md#strokestates-in-rowerjs

Benefit is that this describes the entire framework, and allows the use of finite state machines to describe the relation states.

Abasz, 2025-03-18:
Thanks! this is helpfull. The idea came from the fact that I started to do this for the Peripherals where appropriate (I did not commit these so far as those are not fully complete) because it makes finding simple mistakes easier .

Abasz, 2025-03-18:
>Commands between the managers are documented in the architecture document, see https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/Architecture.md#command-flow

Is it possible you are missing this: `setIndoorBikeSimulationParameters`

Abasz, 2025-03-18:
I noticed that the sessionType casings are inconsistent between the server and the GUI. 

in the dashbaordMetrics.js file its lower case (apart from Calories which is upper case). Also in the docs its lower case. But in the backend its all upper case.

I think these string types should be converted to "enums" and then we can access them like properties and they are always the same.

```js
const SessionType = {
Distance: "distance",
JustRow: "justrow"
}
```

JaapvanEkris, 2025-03-19:
> I noticed that the sessionType casings are inconsistent between the server and the GUI.
> 
> in the dashbaordMetrics.js file its lower case (apart from Calories which is upper case). Also in the docs its lower case. But in the backend its all upper case.

Where in the backend it is uppercase? The parser (see https://github.com/JaapvanEkris/openrowingmonitor/blob/main/app/engine/utils/workoutSegment.js) is in lowercase. That is the only one that really matters.

> I think these string types should be converted to "enums" and then we can access them like properties and they are always the same.
> 
> const SessionType = {
> Distance: "distance",
> JustRow: "justrow"
> }

In the end, there will be one place parsing them in the engine, and several generating them. I don't want to complicate this too much as not much maintenance is expected once its working.

JaapvanEkris, 2025-03-19:
> Where in the backend it is uppercase?

I grepped them. This is an error that is cut-copy-pasted to all inits of the metrics-object. In some obscure situations the initial metrics broadcast comes too late (as the BLE stack delays startup), requiring this init to prevent a crash due to missing metrics.

Fixxing this by introducing an enum is just overkill.

Abasz, 2025-03-19:
Yes, these are in the BLE Stack. I will fix this, this is totaly fine, I just wanted to know which is the right one.

https://github.com/JaapvanEkris/openrowingmonitor/blob/28fa9f4962f130d6482aaed5d7a9f80d96e4bcc1/app/peripherals/ble/pm5/characteristic/GeneralStatus.js#L45

Also int the GUI:
https://github.com/JaapvanEkris/openrowingmonitor/blob/28fa9f4962f130d6482aaed5d7a9f80d96e4bcc1/app/client/store/dashboardMetrics.js#L57

JaapvanEkris, 2025-03-19:
> Yes, these are in the BLE Stack. I will fix this, this is totaly fine, I just wanted to know which is the right one.
> 
> [openrowingmonitor/app/peripherals/ble/pm5/characteristic/GeneralStatus.js](https://github.com/JaapvanEkris/openrowingmonitor/blob/28fa9f4962f130d6482aaed5d7a9f80d96e4bcc1/app/peripherals/ble/pm5/characteristic/GeneralStatus.js#L45)
> 
> Line 45 in [28fa9f4](/JaapvanEkris/openrowingmonitor/commit/28fa9f4962f130d6482aaed5d7a9f80d96e4bcc1)
> 
>  bufferBuilder.writeUInt8(data.sessiontype === 'Distance' ? 2 : (data.sessiontype === 'Time' ? 4 : 0)) 

Good catch! My bad.

> Also int the GUI:
>
> [openrowingmonitor/app/client/store/dashboardMetrics.js](https://github.com/JaapvanEkris/openrowingmonitor/blob/28fa9f4962f130d6482aaed5d7a9f80d96e4bcc1/app/client/store/dashboardMetrics.js#L57)
> 
> Line 57 in [28fa9f4](/JaapvanEkris/openrowingmonitor/commit/28fa9f4962f130d6482aaed5d7a9f80d96e4bcc1)
> 
>  const calories = metrics?.sessiontype === 'Calories' ? Math.max(metrics?.intervalTargetCalories - metrics?.intervalLinearCalories, 0) : metrics?.totalCalories

Yeah, that is a tricky one. Still trying to find a way to do that well as Calories aren't easy to define as continuous metric. I have some homework to do there....

Abasz, 2025-03-19:
> Fixing this by introducing an enum is just overkill.
> In the end, there will be one place parsing them in the engine, and several generating them. I don't want to complicate this too much as not much maintenance is expected once its working.

the thing is that the PM5 spec uses enums and parses them based on their int value. So for the PM5 these strings would need to be converted to numbers for instance for the woroutType. 

Changing these to "enum" like objects (where property is the string and the value is a number from 0) is very easy, just need to do a search in VSCode and replace them. What I am trying to say is that for the CSAFE to work, I will need to define a large amount of enums as the parsing those is simpler. I can define the workoutType enum within the peripherals and then I can use square bracket object notation (like sessionTypeEnum[data.sessiontype]) so we avoid a big switch statement for parsing

This is not the only one actually.

```cpp
typedef enum {
WORKOUTTYPE_JUSTROW_NOSPLITS,
/**< JustRow, no splits (0). */
WORKOUTTYPE_JUSTROW_SPLITS,
/**< JustRow, splits (1). */
WORKOUTTYPE_FIXEDDIST_NOSPLITS,
/**< Fixed distance, no splits (2). */
WORKOUTTYPE_FIXEDDIST_SPLITS,
/**< Fixed distance, splits (3). */
WORKOUTTYPE_FIXEDTIME_NOSPLITS,
/**< Fixed time, no splits (4). */
WORKOUTTYPE_FIXEDTIME_SPLITS,
/**< Fixed time, splits (5). */
WORKOUTTYPE_FIXEDTIME_INTERVAL,
/**< Fixed time interval (6). */
WORKOUTTYPE_FIXEDDIST_INTERVAL,
/**< Fixed distance interval (7). */
WORKOUTTYPE_VARIABLE_INTERVAL,
/**< Variable interval (8). */
WORKOUTTYPE_VARIABLE_UNDEFINEDREST_INTERVAL,
/**< Variable interval, undefined rest (9).
WORKOUTTYPE_FIXEDCALORIE_SPLITS,
/**< Fixed calorie, splits (10). */
WORKOUTTYPE_FIXEDWATTMINUTE_SPLITS,
/**< Fixed watt-minute, splits (11). */
WORKOUTTYPE_FIXEDCALS_INTERVAL,
/**< Fixed calorie interval (12). */
WORKOUTTYPE_NUM
/**< Number of workout types (13). */
} OBJ_WORKOUTTYPE_T;
```

JaapvanEkris, 2025-03-19:
> the thing is that the PM5 spec uses enums and parses them based on their int value. So for the PM5 these strings would need to be converted to numbers for instance for the woroutType.

But Garmin doesn't, and we use their structure as a basis as it is the most flexible, matches with fit-files we produce and is used as a workout-file we will import later on (both Garmin and RowsAndAll provide their workouts in fit format).

The session manager is organised this way as well: it has a workoutSegment module that has several instances: one that manages the interval, and one that manages the split. They are independent and the session manager just keeps them in sync when needed. The split essentially is a workoutsegment within the interval that is handed over, but from there on they lead their own independent lives. Thus we are much more flexible than C2 and Garmin are: ORM allows distance based intervals with time based splits, and vice versa. So their fixed structure will never fit our more flexible one.

The MQTT interface also receiving this format is extremely simple, with just a few simple choices. In the overloaded C2 approach you have to have a guide for selecting the right workouttypewithin the enum based on a combination of selections. In our approach a simple Home Automation system can always do it based on user input. I'm currently playing with HA bridge, Alexa and Domticz to see if I can set my trainingparameters via voice command. You don't want much complexity there.

Same goes for the webgui.

> Changing these to "enum" like objects (where property is the string and the value is a number from 0) is very easy, just need to do a search in VSCode and replace them. What I am trying to say is that for the CSAFE to work, I will need to define a large amount of enums as the parsing those is simpler. I can define the workoutType enum within the peripherals and then I can use square bracket object notation (like sessionTypeEnum[data.sessiontype]) so we avoid a big switch statement for parsing
> 
> This is not the only one actually.
> 
> typedef enum {
> WORKOUTTYPE_JUSTROW_NOSPLITS,
> /**< JustRow, no splits (0). */
> WORKOUTTYPE_JUSTROW_SPLITS,
> /**< JustRow, splits (1). */
> WORKOUTTYPE_FIXEDDIST_NOSPLITS,
> /**< Fixed distance, no splits (2). */
> WORKOUTTYPE_FIXEDDIST_SPLITS,
> /**< Fixed distance, splits (3). */
> WORKOUTTYPE_FIXEDTIME_NOSPLITS,
> /**< Fixed time, no splits (4). */
> WORKOUTTYPE_FIXEDTIME_SPLITS,
> /**< Fixed time, splits (5). */
> WORKOUTTYPE_FIXEDTIME_INTERVAL,
> /**< Fixed time interval (6). */
> WORKOUTTYPE_FIXEDDIST_INTERVAL,
> /**< Fixed distance interval (7). */
> WORKOUTTYPE_VARIABLE_INTERVAL,
> /**< Variable interval (8). */
> WORKOUTTYPE_VARIABLE_UNDEFINEDREST_INTERVAL,
> /**< Variable interval, undefined rest (9).
> WORKOUTTYPE_FIXEDCALORIE_SPLITS,
> /**< Fixed calorie, splits (10). */
> WORKOUTTYPE_FIXEDWATTMINUTE_SPLITS,
> /**< Fixed watt-minute, splits (11). */
> WORKOUTTYPE_FIXEDCALS_INTERVAL,
> /**< Fixed calorie interval (12). */
> WORKOUTTYPE_NUM
> /**< Number of workout types (13). */
> } OBJ_WORKOUTTYPE_T;

Those enums do not match up with our internal format anyway, so we have to parse them at the interface to our internal format. C2 has an extremely bad habit of overloading this type of parameters, which is just messy programming. For example, WORKOUTTYPE_FIXEDDIST_NOSPLITS and WORKOUTTYPE_FIXEDDIST_SPLITS map to the same top workout structure (a fixed distance workout), and an absent/present split object.

I'll make a mapping between the two.

Abasz, 2025-03-19:
That is clear. And i actually fully agree with you. the only issue is that if one wants to have the workout received from a PM5 compliant device all the PM5 types need to be mapped to our internal session logic. And actually, its not just the workouttype but whole structural logic how that workout is built. So technically there is a need for a parser of the CSAFE commands and structure and then a translator that translates these back and forth between ORM and the sender. Because the sending end will be using ots own structure (and because we are not just receiving but sending statuses and split and interval data back as part the session, hence the back and forth translation/mapping). 

I can do the parser (probably), though the spec is a bit messy, so it will take more time, I see messages from ergData that does not seem to comply (e.g. start with 0x01 bytes and only then comes the Start flag byte of 0xf0/0xf1. But for the translation I leave it to you then.

JaapvanEkris, 2025-03-19:
> WORKOUTTYPE_JUSTROW_NOSPLITS,  /**< JustRow, no splits (0). */

Essentially, a just row session:
```js
[
  {
    "type": "justrow"
  }
]
```

> WORKOUTTYPE_JUSTROW_SPLITS, /**< JustRow, splits (1). */

Essentially, for a just row session, 500m split:
```js
[
  {
    "type": "justrow",
    "split": {
      "type": "distance",
      "targetDistance": "500"
    }
  }
]
]
```

> WORKOUTTYPE_FIXEDDIST_NOSPLITS, /**< Fixed distance, no splits (2). */

Essentially, for a 5K:
```js
[
  {
    "type": "distance",
    "targetDistance": "5000"
  }
]
```

> WORKOUTTYPE_FIXEDDIST_SPLITS, /**< Fixed distance, splits (3). */

Essentially, for a 5K, 500m split:
```js
[
  {
    "type": "distance",
    "targetDistance": "5000",
    "split": {
      "type": "distance",
      "targetDistance": "500"
    }
  }
]
]
```

> WORKOUTTYPE_FIXEDTIME_NOSPLITS, /**< Fixed time, no splits (4). */

A 2 minute session
```js
[
  {
    "type": "time",
    "targetTime": "120"
  }
]
```

> WORKOUTTYPE_FIXEDTIME_SPLITS, /**< Fixed time, splits (5). */

A 2 minute session, with 30 sec splits
```js
[
  {
    "type": "time",
    "targetTime": "120",
    "split": {
      "type": "time",
      "targetTime": "30"
    }
  }
]
```

> WORKOUTTYPE_FIXEDTIME_INTERVAL, /**< Fixed time interval (6). */

Simply add an adittional time-based workout object to the array.

> WORKOUTTYPE_FIXEDDIST_INTERVAL, /**< Fixed distance interval (7). */

Simply add an adittional time-based workout object to the array.

> WORKOUTTYPE_VARIABLE_INTERVAL, /**< Variable interval (8). */

Simply add an adittional time-based workout object to the array.

> WORKOUTTYPE_VARIABLE_UNDEFINEDREST_INTERVAL, /**< Variable interval, undefined rest (9).

A rest interval as such isn't available as such, as they consider it a low intensity interval. One could map it to our 'rest' interval, which stops all metrics (end time required).

> WORKOUTTYPE_FIXEDCALORIE_SPLITS, /**< Fixed calorie, splits (10). */

Not supported yet, map to 'justrow'

> WORKOUTTYPE_FIXEDWATTMINUTE_SPLITS, /**< Fixed watt-minute, splits (11). */

This is a workout target (i.e. minimal pace/power) combinaed with a time based interval. Not supported yet.

> WORKOUTTYPE_FIXEDCALS_INTERVAL, /**< Fixed calorie interval (12). */

Not supported yet, map to 'justrow'

JaapvanEkris, 2025-03-19:
> That is clear. And i actually fully agree with you. the only issue is that if one wants to have the workout received from a PM5 compliant device all the PM5 types need to be mapped to our internal [...]

Well, my impression is that most apps blindly assume the PM5 will execute the plan as proposed, and just see the results coming by. So when receiving a plan we need to do some parsing, but in reporting we can map quite loosely (we already do with the session type). So we need to understand the workout plan, but reporting back can be based on our internal mapping and a general direction (like the bug you found in GeneralStatus).

> Because the sending end will be using ots own structure (and because we are not just receiving but sending statusses and split and interval data back as part the session, hence the back and forth translation/mapping).

I have the impression app developers don't have a very stateful approach. When we send an interval end report, they will assume it is valid and process the data as such (record it, present it). We do the same in the fit recorder: the workout plan and the results are in the same file, but that is about it. When the engine indicates a new interval, we'll record it as such without looking to the plan. This is because when people do an unplanned stop, all interval and split numbering will be off. If the plan and result are too tightly coupled, that will derail a lot of communication.

> I can do the parser (probably), though the spec is a bit messy, so it will take more time, I see messages from ergData that does not seem to comply (e.g. start with 0x01 bytes and only then comes the Start flag byte of 0xf0/0xf1. But for the translation I leave it to you then.

No problem!

closed 2025-07-06 by JaapvanEkris