JaapvanEkris/openrowingmonitor #97: RPI 3B installed fails

Frits1980, 2025-04-13:
Hi,

Just grabbed an old RPI 3B to test this. Installed Bookworm 64 bit lite on it. And ran the install script.
Got this error but don't know how to solve it.

[image]

JaapvanEkris, 2025-04-14:
Could you try installing the new-ble-api branch (installation manual describes what to do)?

It's PM5 interface is still in development, and it will take a while before we merge it into main, but Abasz replaced the Bluetooth stack this complaint is about.

Frits1980, 2025-04-14:
Hi Jaap,

Thank you for your reaction. I've tried your sugestion but I still got the errors. Herewith the complete log file.

[log.txt](https://github.com/user-attachments/files/19731888/log.txt)

JaapvanEkris, 2025-04-14:
> Hi Jaap,
> 
> Thank you for your reaction. I've tried your sugestion but I still got the errors. Herewith the complete log file.
> 
> [log.txt](https://github.com/user-attachments/files/19731888/log.txt)

Looks like you are still installing the current main. Installing an alternative branch can be done via:

```zsh
wget https://raw.githubusercontent.com/jaapvanekris/openrowingmonitor/HEAD/install/install.sh
```

Followed by opening the downloaded file in a text editor (nano in this case):

```zsh
sudo nano install.sh
```

Here, look for the line

```zsh
BRANCH="main"
```

And change the name of the branch into "new-ble-api". Save the file. You can now install the branch by running

```zsh
sudo /bin/bash ./install.sh
```

Just answer the questions from the script and OpenRowingMonitor will be installed for you completely.

Frits1980, 2025-04-14:
Thank you again for the quick reply. I followed your instructions and tried again. Unfortunately with the same result. Again here is the log file.

[log.txt](https://github.com/user-attachments/files/19732403/log.txt)

Abasz, 2025-04-14:
new-ble-api branch do not depend on @abandonware BLE. So you must be installing the wrong branch. can you please delete the /opt/openrowingmonitor folder and rerun the edited install script (i.e. where you change the branch name to BRANCH="new-ble-api"

Frits1980, 2025-04-14:
Hi Abasz,

Thank you for your guidance. This did the trick. It works perfect now.

Kind regards, Frits

closed 2025-04-14 by Frits1980

JaapvanEkris, 2025-04-15:
@Frits1980 Stupid question perhaps: why is it asking for the abandonware version anyways (current main is using stoprecent's branch)? Did you first install laberning's version first?

Please explain, as we want to move away from this implementation, and we seem to have missed a spot.

Frits1980, 2025-04-15:
I did install laberning version first. Maybe add an option to keep or remove old installation when running the install script?

Anyway it runs very nice now on my rpi3b with the original sensor of the first degree fitness s500 plugged in gpio 26 and ground. Didn’t even need to remove the plug from the wires 😁

JaapvanEkris, 2025-04-16:
> I did install laberning version first. Maybe add an option to keep or remove old installation when running the install script?

Ah, that explains it. Yeah, perhaps offering a 'clean install' option when the directory isn't empty is a good addition.

> Anyway it runs very nice now on my rpi3b with the original sensor of the first degree fitness s500 plugged in gpio 26 and ground. Didn’t even need to remove the plug from the wires 😁

Yeah, that is the easiest approach. Have fun with it!

JaapvanEkris, 2025-04-16:
> Thank you for all the hard work you put in to this project.

You are more than welcome!

> I do have one question though. A little off topic, but how would I go about when I want to use this rower with my wife. Obviously she has an other Strava account. How to easily switch?

This indeed is a problem. It is even in the backlog. In the backend we are inching towards multi-user setups (newer uploaders are already part of the user object, and as soon as I understand how the Strava code works, that will be mirated to that setup as well), but we need a front-end developer to help us implement it. There are two alternative approaches:
* Use the EXR game. We integrate well with them. Their familyplan allows use of three accounts. They do the Strava uploading and add pretty pictures and a map. My wife loves it, as do I, so it keeps me rowing. But not everybodies cup of tea, or it might not be in your budget.
* Make a shell script that watches the directory where OpenRowingMonitor places its files, and upload them to the right Strava account from there (there are quite some tools out there that allow CLI uploading). I can recognize my rowing from my wife's by its pace (we structurally report to Syslog) and the serial of the HRM monitor used (also in syslog).

Frits1980, 2025-04-16:
Ok, maybe I can help with the frontend part. I used to be one. About 15 years ago though. But maybe I can still work it out for this project.

JaapvanEkris, 2025-04-16:
> Ok, maybe I can help with the frontend part. I used to be one. About 15 years ago though. But maybe I can still work it out for this project.

That would be a great addition to the team! We suck at frontend development. Definitely needs some love.

I described the inner workings of ORM a bit in the [architecture document](https://github.com/JaapvanEkris/openrowingmonitor/blob/new-ble-api/docs/Architecture.md), and we are reaching the odd point where we can program a workout via voice control (Alexa via MQTT) and App (ErgZone) but not via the GUI. It is getting a bit embarrassing I guess :)

As it is intended to be on simple RPi's and we drain precious CPU cycles from the monitoring of GPIO-pins, we keep it as light as possible. But there is a lot of room for improvement.

Frits1980, 2025-04-16:
I'd love to help out where I can. Please send me the document, then I can get an idea of what is what. I couldn't find a way to message each other other then publicly providing my email. So herewith: info@fgpm.nl