JaapvanEkris/openrowingmonitor #114: Unit webbrowserkiosk.service could not be found

syscleaner, 2025-05-24:
Hi There,

I was running through the manual. 

on the step checking webbrowserkiosk I got the bnelow error message

Unit webbrowserkiosk.service could not be found.

Also I connect small hdmi screen to gui overthere but it is just sitting on command prompt. 

any suggestions ? 

Thanks

Gordon-Shumway2, 2025-05-24:
Which manual are you using? Please state the web address where you downloaded the manual.
Can you open a web browser (not on the pi but on some other PC) and use that to open the gui at
  
 http://<your pi's IP>/

Replace "<your pi's IP>" in the line above with your pi's IP. If the IP is unknown check on your router for the ip address of your pi.

JaapvanEkris, 2025-05-24:
> on the step checking webbrowserkiosk I got the bnelow error message
> 
> Unit webbrowserkiosk.service could not be found.

What version did you install (I assume the main branch found in this repository), and did you get any error messages during the install process?

syscleaner, 2025-05-25:
Hi There,

Yes just follow directions from this repository. 

I didn't see any particular error message but I can redo every thing. Last I was getting same error message I changed my SD card and it improved install quite alot. I can try another card. 

Also this is the touch screen that I want to use but I can't make it goin to UI. Even I enter password it just sits in command line. 

[image]

Gordon-Shumway2, 2025-05-25:
The GUI is a "graphical user interface". What you are looking at is a text based interface - the prompt. You will not find the GUI there. 
You will find the GUI (if ORM is up and running) by starting a browser (Firefox, Chrome, etc.) and opening it at

    http://192.168.1.66/

Regards, 
Michael

JaapvanEkris, 2025-05-31:
@syscleaner any progress in this case?

Jaap

syscleaner, 2025-05-31:
Hi there I was just trying with new card and new Pi OS 11 bullseye. 

Before I was not checking the version and I just said yes to ever install item even it says this OS not tested. 

Now I it is much worse. 

sero@rowingmonitor:~ $ sudo setcap cap_net_bind_service,cap_net_raw=+eip $(eval readlink -f `which node`)
sero@rowingmonitor:~ $ sudo systemctl status openrowingmonitor
**Unit openrowingmonitor.service could not be found.**
sero@rowingmonitor:~ $ sudo journalctl -u openrowingmonitor
-- Journal begins at Tue 2025-05-06 20:32:40 +07, ends at Sat 2025-05-31 21:06:32 +07. --
-- No entries --
sero@rowingmonitor:~ $ sudo systemctl status webbrowserkiosk
**Unit webbrowserkiosk.service could not be found.**
sero@rowingmonitor:~ $

JaapvanEkris, 2025-05-31:
Why not use the current OS version, Bookworm?

For most tests I do a clean install and never encounter any issues.

syscleaner, 2025-06-01:
Well new card, new card adapter. 

Now I killed Pi 

I think some how it is overvolt. 

It was okey with that power source my previous attempts but seems my luck not going well. 

[image]

I will trry with an ESP board or aurdunio if available.

JaapvanEkris, 2025-06-11:
Any progress on this issue?

JaapvanEkris, 2025-06-30:
As I can't reproduce this issue on any test devices, and you haven't responded to requests for a month, I'll assume this issue is resolved.

If this isn't the case, please open a new issue.

closed 2025-06-30 by JaapvanEkris