JaapvanEkris/openrowingmonitor #115: Unplanned pauses aren't handled well via PM5

JaapvanEkris, 2025-05-24:
The PM5 emulator  in the 0.9.6 development branch does not handle unplanned well yet. Here, a key difference between the OpenRowingMonitor and the PM5 shows itself, and requires some tricks to manage well. With planned rest periods, the PM5 actually reports the pause as being part of the previous active split/interval. We can control this behaviour well as long is it is a planned pause, and it works pretty flawless at the moment.

When a rest is unplanned, we see a disconnect between the plan on ErgZone and ORM: ErgZone assumes the split/interval is closed and starts the new one way too early, leading to weird effects as planned schedule of ORM is out of sync with ErgZone. We can easily distinguish unplanned rest periods from planned ones (as with planned intervals the above interval should be of type 'rest' and not a 'time' or 'distance' type).

However, communicating this without causing total chaos on the ErgZone end is more challenging. In essence, I guess, we need to merge three intervals into one, which is challenging. So we got some work to do.

Please be aware of this when testing.

JaapvanEkris, 2025-06-01:
I think I have an approach that might work, but it adds quite some complexity. It is a two-part solution:
* The sessionManager will make sure the split is concluded as planned. So any remaining time/distance is set as target for the next split (unless the interval boundary is crossed). This makes sure an unplanned rest doesn't disturb the interval lengths (which throws off ErgZone);
* The PmRowingService will glue the pre-pause-split, pause-split and post-pause-split into a single split. In order to do so, it must keep a "running sum" across two splits and keep the totals when appropriate. Tricky thing is that there is nothing stopping a user from having two or three breaks in the same split, followed by a planned rest interval.

Abasz, 2025-06-02:
While I understand that the pause behaviour is driven by PM5 to some extent, the part I dont really understand is why we are trying to adhere it so strictly. 

In any decent workout manager (e.g. garmin) if you take a non-planned pause during a workout (e.g. when you should have been rowing) that pause is not considered as pause from a timer's and metrics perspective in the sense that it stops. What I mean is that even if the I stop my watch will keep counting time and my average split/interval time will decrease.

So I wonder if we could cut this Gordian knot by saying if you are in an interval or split and you take a non-planned pause we dont pause?

to be fair this is much closer to real life: If you are on a competition and you stop for a minute the timer will not stop and nothing will suddenly pause and start from where you left off when you restart. 

Of course we should give the user to trigger uploads if they decide to exit the workout but we have reset button on the gui. So If they pause with the intention of stopping the workout they can explicitly do this.

JaapvanEkris, 2025-06-02:
> While I understand that the pause behaviour is driven by PM5 to some extent, the part I dont really understand is why we are trying to adhere it so strictly.

The key issue is that our core timer is the sum of the currentDt's, not the system time. So if the flywheel becomes too slow and we pause, we actually switch between these measurement systems (as during a pause the flywheel might actually stop). This works because we distinguish between the two (and system time fluctuates too much to be precise, where CPU-ticks have less issues).

> So I wonder if we could cut this Gordian knot by saying if you are in an interval or split and you take a non-planned pause we dont pause?

That is what C2 is doing. I now have a version on my test system that simply stacks split metrics from pauses/non-pause segments until the split end is reached. I found that our flags in GeneralMessage are also wrong, so I'll see how it all behaves tonight.

JaapvanEkris, 2025-06-02:
> indeed... I keep forgetting this. There is no way around switching between the two count system... we must do two tracking actually. when flywheel stops and normal time stops we need to record a timestamp and when restarts we need to add back. 

The `SessionManager.js` and `WorkoutSegment.js` do this, and I don't see too many issues in testing (it already was part of the first 0.9.6 builds). Currently it runs OKish: sums are correct, etc. So the carry over and merge seem to work well.

The key issue is those darn flags. Something is triggering ErgZone to switch to the next split at the start of the unplanned pause, and thus the session is cut short. I made Excel spreadsheets so that I can load a BLE-trace or message-dump and compare the behaviour. See where it bahaves differently.

JaapvanEkris, 2025-06-05:
It looks like I found a working solution. What remains is:
* making all metrics use a consistent scope (split with pause included, workout data with some consistency),
* documenting our choices, because in six months I'll have forgotten about all these traps and weird choices from C2,
* check if all other use cases survived,
* testing the combinations of unplanned pauses with these use cses.

If that all works, I'll start endurance testing and we have a release mid to late june.