JaapvanEkris/openrowingmonitor #118: 0.9.6 ErgZone Workout Push Not Working

jryd2000, 2025-05-27:
When I try to push an interval workout from ErgZone (or ErgData), the intervals are not create in open rower.  Looking at the logs it appears that a crash occurs and the openrower service restarts.  

Steps to Reproduce:
1.) Restart the openrower service with bluetoothMode set to PM5.
2.) Open ErgZone and create a HIIT interval with a 5 minute time interval with a 5 minute rest, and then a 4 minute interval and 4 minute rest, down to a 1 minute interval with a 1 minute rest (Total workout of 30mins).
3.) Press the Scan For Monitor in ErgZone and select the PM5 device that is found and Press Connect.
4.) I get a Connecting message and then a Sending Workout, Wait message and then it completes and says Row to Start!

However the log file indicates that a crash occurred and has restarted.

[Log.txt](https://github.com/user-attachments/files/20457519/Log.txt)

JaapvanEkris, 2025-05-27:
Hmmm, Have seen this one from ErgData, not from ErgZone yet. But when I configure your workoutplan in ErgZone, it will indeed crash the app.

The relevant part of the log is:
```sh
May 27 08:52:17 rowingmonitor npm[1803]: PM5 Control frame is complete: [
May 27 08:52:17 rowingmonitor npm[1803]:   '0xf1', '0x76', '0x60', '0x18', '0x01', '0x00', '0x01',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x01', '0x08', '0x17', '0x01', '0x00', '0x03', '0x05',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x00', '0x00', '0x00', '0x75', '0x30', '0x04', '0x02',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x01', '0x2c', '0x14', '0x01', '0x01', '0x18', '0x01',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x01', '0x17', '0x01', '0x00', '0x03', '0x05', '0x00',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x00', '0x00', '0x5d', '0xc0', '0x04', '0x02', '0x00',
May 27 08:52:17 rowingmonitor npm[1803]:   '0xf3', '0x00', '0x14', '0x01', '0x01', '0x18', '0x01',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x02', '0x17', '0x01', '0x00', '0x03', '0x05', '0x00',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x00', '0x00', '0x46', '0x50', '0x04', '0x02', '0x00',
May 27 08:52:17 rowingmonitor npm[1803]:   '0xb4', '0x14', '0x01', '0x01', '0x18', '0x01', '0x03',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x17', '0x01', '0x00', '0x03', '0x05', '0x00', '0x00',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x00', '0x2e', '0xe0', '0x04', '0x02', '0x00', '0x78',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x14', '0x01', '0x01', '0x18', '0x01', '0x04', '0x17',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x01', '0x00', '0x03', '0x05', '0x00', '0x00', '0x00',
May 27 08:52:17 rowingmonitor npm[1803]:   '0x17', '0x70',
May 27 08:52:17 rowingmonitor npm[1803]:   ... 2 more items
May 27 08:52:17 rowingmonitor npm[1803]: ]
May 27 08:52:17 rowingmonitor npm[1803]: InvalidFrameError: Checksum does not match. 0xf1,0x76,0x60,0x18,0x01,0x00,0x01,0x01,0x08,0x17,0x01,0x00,0x03,0x05,0x00,0x00,0x00,0x75,0x30,0x04,0x02,0x01,0x2c,0x14,0x01,0x01,0x18,0x01,0x01,0x17,0>
May 27 08:52:17 rowingmonitor npm[1803]:     at new CsafeRequestFrame (file:///opt/openrowingmonitor/app/peripherals/ble/pm5/csafe-service/CsafeRequestFrame.js:71:13)
May 27 08:52:17 rowingmonitor npm[1803]:     at CsafeManagerService.processCommand (file:///opt/openrowingmonitor/app/peripherals/ble/pm5/csafe-service/CsafeManagerService.js:42:24)
May 27 08:52:17 rowingmonitor npm[1803]:     at Object.onWrite (file:///opt/openrowingmonitor/app/peripherals/ble/pm5/control-service/ControlReceiveCharacteristic.js:68:37)
May 27 08:52:17 rowingmonitor npm[1803]:     at Object.write (/opt/openrowingmonitor/node_modules/ble-host/lib/internal/gatt.js:637:12)
May 27 08:52:17 rowingmonitor npm[1803]:     at /opt/openrowingmonitor/node_modules/ble-host/lib/internal/gatt.js:1486:12
May 27 08:52:17 rowingmonitor npm[1803]:     at Object.authorizeWrite (/opt/openrowingmonitor/node_modules/ble-host/lib/internal/gatt.js:671:8)
May 27 08:52:17 rowingmonitor npm[1803]:     at Object.<anonymous> (/opt/openrowingmonitor/node_modules/ble-host/lib/internal/gatt.js:1473:10)
May 27 08:52:17 rowingmonitor npm[1803]:     at AclConnection.<anonymous> (/opt/openrowingmonitor/node_modules/ble-host/lib/ble-manager.js:1703:14)
May 27 08:52:17 rowingmonitor npm[1803]:     at AclConnection.emit (node:events:524:28)
May 27 08:52:17 rowingmonitor npm[1803]:     at HciSocket.onData (/opt/openrowingmonitor/node_modules/ble-host/lib/internal/adapter.js:1231:16)
```
@Abasz any idea what causes this?

JaapvanEkris, 2025-05-27:
> down to a 1 minute interval with a 1 minute rest (Total workout of 30mins)

Please note that OpenRowingMonitor will remove the last `rest` split (as it likes to complete the workout decently and start uploads etc..). We can't handle rests intervals at the final interval, as we can only conclude a rest interval has ended by detecting a new drive. ErgData will get the "End of Workout" message and will respect it on their side: it will skip the last rest minute.

PS: Thank you for testing OpenRowingMonitor so thoroughly!

jryd2000, 2025-05-28:
It appears to be working for me now.  Now when I push the workout, I get the timer countdown.  I haven't actually rowed the whole workout yet.  I will try it tomorrow morning.

JaapvanEkris, 2025-05-30:
> I updated the 0.9.6 branch and gave it another try. Same result. Updated log file.
> 
> [log.txt](https://github.com/user-attachments/files/20512544/log.txt)

Interesting. It is recieving the workout, but you are not recognizing it as such:
```sh
May 29 19:16:40 rowingmonitor npm[1481]: SessionManager: Workout plan recieved with 1 interval(s)
May 29 19:16:40 rowingmonitor npm[1481]:   Workout parser, recognised justrow interval/split
May 29 19:16:40 rowingmonitor npm[1481]: Activating interval settings for interval 1 of 1
May 29 19:16:40 rowingmonitor npm[1481]:   Workout parser, recognised justrow interval/split
May 29 19:16:40 rowingmonitor npm[1481]: Activating split settings for split 1
May 29 19:16:40 rowingmonitor npm[1481]:   Workout parser, recognised time interval/split, 480 seconds
```
This is because the type used can't be mapped perfectly. Still this workout plan?
> workout with 3 intervals with 8 minutes of rowing and 2 minutes of rest per interval 

HOW did you program them in ErgZone? As a mixed interval session?

jryd2000, 2025-05-30:
In ErgZone, the only option I have is to manually add intervals.  In the App, I press the plus button to Add Workout, then I can only choose the type of workout (rowing, cycling, etc) and then you have to press New Interval.  At that point you enter the Type (time) and enter the 8 minutes and 2 minutes rest and press Close.  Then you have to press New Interval again, and it remembers the last interval so you just press close again.  Then you just do it a third time and press Save.

You were correct that if I change at least one of the times (make the last interval 8:01) it works perfectly fine (and that is what I am doing as a workaround for now).  

I see in Ergdata they have the different types of workouts for Constant Intervals and Variable Intervals.  If I do Constant intervals and choose 8mins and 2 minutes rest, it does seem to work.  Likewise, if I do Variable Intervals and build 3 explicit intervals with 8 minutes rowing and 2 minutes rest it also works.  But I'm not sure if the latter is in the background converting to a Constant Intervals workout.

JaapvanEkris, 2025-05-30:
Thanks for your good description. Extremely helpfull as a I can repeat it now. It seems that in the background ErgZone translates this into a `WORKOUTTYPE_FIXEDTIME_INTERVAL`, with an interval of 8:00 and rest of 2:00. This type of interval is endless, so it keeps repeating to eternity. When I add an extra interval, I get the exact same commands and parameters, so it seems that ErgZone is optimizing here.

I do see we can get a lot closer to the PM5's response by essentially making a workoutplan with these intervals repeated as much as we can. That would make a better approaximation and trigger a visual change, which is needed to understand the programming has succeeded.

But please realise, the "just three intervals" is getting lost in translation here, unfortunatly on ErgZone's side. So if your wife is totally demolished during a workout, she probably lost count, but we are not the ones to blame here.... :)

Abasz, 2025-05-30:
Jaap, I dont understand this looping issue. How is that ErgZone sends infinite number of intervalls/splits (sorry I dont really get the difference still :):))? Isnt this a bug on their side? How would a PM5 deal with this? Would it program 50 splits, essentially making a 500 minutes workout? How is that logical in any workout creation sense? :D

JaapvanEkris, 2025-05-30:
> Jaap, I dont understand this looping issue. How is that ErgZone sends infinite number of intervalls/splits (sorry I dont really get the difference still :):))? Isnt this a bug on their side? How would a PM5 deal with this? 

It actually is the specific workout type they send: 'WORKOUTTYPE_FIXEDTIME_INTERVAL'. It is an interval type that ErgData sends along with a single interval specification (with a rest if needed). It is interpreted by the PM5 to repeat forever (see [this concept2 video](https://youtu.be/ZCdY-874WlM). As a PM5 has an upper limit of 50 splits/intervals, I assume this is the case here as well.

The issue is that ErgZone is optimizing behind the scenes:
* if you make the described workout of jrdy2000's wife, it will translate it to a 'WORKOUTTYPE_FIXEDTIME_INTERVAL'.
* If you skip the rests, it will make a single time interval with splits of the length of your intervals.
* If you take jrdy2000's wife workout and add a single second to the last interval, it will become a varied interval workout.

Consistency clearly isn't a consideration. Workouts 1 and 3 will result in a timer showing 8 minutes, option 2 will show 24 minutes. Workout 1 will go on forever (or to the 50th split, whatever that might be, as it is extremely ambiguous in the C2 world).

> Would it program 50 splits, essentially making a 500 minutes workout? How is that logical in any workout creation sense? :D

You keep on going until you drop? It is C2's invention, not ours...

Abasz, 2025-05-30:
Omg... The person who invented this structure has probably never in his/her life seen a structured workout... The whole pint is that it is predefined defined. 

Alternatively they may only did negative split interval with penalty meaning that if the improvement of time (or distance) is not sufficient must be repeated, in that case this is handy :):) .

jryd2000, 2025-05-30:
That seems to have fixed it.  I can push variable and fixed interval workouts from both ErgZone and ErgData now.