JaapvanEkris/openrowingmonitor #120: 0.9.6 ErgData Interval Data Distance 0

jryd2000, 2025-05-31:
Now that pushing workouts is working, I tried to a Variable Interval workout in ErgData with 3 intervals with 8 minutes of rowing and 2 minutes of rest.  By an large everything worked fine with one exception.  As I was completing intervals I would see that the previous interval would have information about the segment including time, stroke rate, HR, and pace.  However the distance was appearing as 0.  As I was rowing the interval it would show a correct value, but it seems like as soon as the interval completed it updated the total distance for the interval to 0.

For the record the FIT file correctly showed each of the interval distances, so it seems to be isolated to the PM5 peripheral.

JaapvanEkris, 2025-05-31:
Please note we do not support ErgData. It is a chaotic beast. Could you check to see what ErgZone is doing?

jryd2000, 2025-05-31:
Understood.  Will try a similar workout on ErgZone for my next workout and see how it does.

jryd2000, 2025-06-04:
I haven't tried this yet myself, but my wife did a workout using ErgZone and it seemed to be reporting the interval distances correctly.  So must just be an ErgData quirk.

closed 2025-06-04 by JaapvanEkris