JaapvanEkris/openrowingmonitor #123: GUI error on interval type

Abasz, 2025-06-03:
In the GUI I get an error that interval.type is not defined. 

[image]

I think this is the line where the type is not defined:
https://github.com/JaapvanEkris/openrowingmonitor/blob/75880996174542f6835d6728630b6e38e0bc209c/app/client/store/dashboardMetrics.js#L21

JaapvanEkris, 2025-06-03:
At startup?

I usually see this error when the Interval object is missing (the missing key often fails silently).

Abasz, 2025-06-03:
Immediately when loading the GUI. always twice actually :)

If this can be undefined and this is intentional I think adding `?` would solve this :)

JaapvanEkris, 2025-06-04:
I've inserted the new init for the client. Could you check if this behaviour is still present (test machine is occupied with testing unplanned pauses).

JaapvanEkris, 2025-06-05:
That is weird, as the init of the webserver includes a interval.type = 'justrow'. In the SessionManager the constructor of the interval guarantees it is set to 'justrow' by default. Looks like we have to dig a bit deeper here (or add the ? as you proposed :))

JaapvanEkris, 2025-06-10:
I've added the ? operator everywhere in the client, to improve its robustness.

Practical note: make sure you do a "sudo npm run build" after the git-pull, as the client is precompiled (why Lars did that, I'll never understand).

Abasz, 2025-06-11:
Yeah I know that it needs recompiling. Its normal for frontends actually. You know many browsers many platforms minifining, boundling  etc.

Anyway you missed a few lines, I added those, pushed the commit.

closed 2025-06-11 by Abasz

Abasz, 2025-06-12:
> It is the part I keep forgetting when working on the front-end :). In C++ you know you must compile.

Actually if you run rollup with -cw it will be in watch mode and recompile on every file change
build:watch": "rollup -cw

`npm run build:watch`