JaapvanEkris/openrowingmonitor #130: firefox memory warning on rpi zero 2w

beavis69, 2025-07-11:
Hello,

On rpi zero 2w, firefox shows a modal box every time at start with this message "It is not recommended to run Firefox on devices with less than 1GB of RAM."

With chromium i was able to disable it with --no-memcheck but with firefox it seems the is no option to disable it.

So i have tried another lightweight browser midori

So i replaced my  webbrowserkiosk.sh with:
```
#!/bin/bash
#
#  Open Rowing Monitor, https://github.com/JaapvanEkris/openrowingmonitor
#
#  Runs the Web Frontend in a Firefox browser in fullscreen kiosk mode
#
xset s off
xset s noblank
xset -dpms
unclutter &
matchbox-window-manager &
# Start Midori in fullscreen mode
midori --display=:0 -e Fullscreen -e Navigationbar -a http://127.0.0.1/?mode=kiosk &
sleep 5
# send F11 to midori to remove the window-bar
xdotool key F11
sleep 365d
```

* https://astian.org/midori-browser/ 
* https://github.com/goastian/midori-desktop

I also replaced openbox with matchbox as matchbox will not try to open menu and printer config stuff (so less warnings in journalctl).

Give it a try, it really use less cpu and much less ram than firefox and chromium.

Hope that helps.

Beavis

JaapvanEkris, 2025-07-11:
> On rpi zero 2w, firefox shows a modal box every time at start with this message "It is not recommended to run Firefox on devices with less than 1GB of RAM."
> 
> With chromium i was able to disable it with --no-memcheck but with firefox it seems there is no option to disable it.

Thanks for thinking along.

> * https://astian.org/midori-browser/
> * https://github.com/goastian/midori-desktop
>
> I also replaced openbox with matchbox as matchbox will not try to open menu and printer config stuff (so less [...]

I will, bit how to install it? I tried 'sudo apt-get install midori' on the 64bit bookworm versiob, but the package isn't known. Any tips here?

beavis69, 2025-07-11:
got mine from raspbian repo:
```
root@rower:~# dpkg --print-architecture
armhf
root@rower:~# cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ bookworm main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
root@rower:~# apt search midori
Sorting... Done
Full Text Search... Done
midori/stable,now 7.0-2.1 armhf [installed]
  fast, lightweight graphical web browser
```

Seems it was removed from debian after bullseyes, but debs are available here : https://astian.org/midori-browser/linux/