JaapvanEkris/openrowingmonitor #132: Strava upload button missing

Ravian18, 2025-07-27:
Ever since I've updated to v0.9.6, the Strava upload button disappeared from the GUI. The [integrations manual](https://github.com/JaapvanEkris/openrowingmonitor/blob/0.9.6/Integrations.md) link as mentioned on [the release page](https://github.com/JaapvanEkris/openrowingmonitor/releases/tag/0.9.6) also returns a 404:
[image]

I was using `stravaClientId` and  `stravaClientSecret` but also already tried replacing it with this:

```js
  strava: {
    allowUpload: true,
    autoUpload: false,
    clientId: '',
    clientSecret: '',
    refreshToken: ''
  }
```

None of the options seem to return the upload button. What am I doing wrong?

Ravian18, 2025-07-27:
Nevermind, I forget to add the `userSettings` object, that was why the button was not showing. I found the instructions [here](https://github.com/JaapvanEkris/openrowingmonitor/blob/main/docs/Integrations.md#strava) btw, if you want to update the release notes :)

JaapvanEkris, 2025-07-28:
Release notes have been updated, thanks for reporting!

commit e7c40873 (2023-04-19): Fix HR battery level not showing (fixes: #132)

Fix the HR battery level not showing on the HR metric tile (ANT and BLE)
For ANT handle case when BatteryLevel is not present only BatteryStatus.
BLE devices do not necessarily send battery level with all HR broadcast so
read the battery level on connect.